sparkfun/SparkFun_MAX3010x_Sensor_Library #5: No Values and LED not working

novel-notion, 2017-11-13:
I am using a MAX301 that I have verified to be working with another library. However the other library was not giving me accurate data so I decided to try this one. The example sketch needed to have a library header file changed from "arduino.h" to "Arduino.h" in order to be able to compile.

After that the code uploaded but the LED on the MAX does not light and I only  get 0 values. I am using an ESP8266.
Any suggestions?

nseidle, 2018-02-15:
Which IC are you using? MAX301 is not one I think exists.

Can you let me know which example you're trying?

Can you let me know which arduino.h line you're talking about? Is it [this one](https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/blob/master/src/spo2_algorithm.cpp#L60)?

Are you illuminating the IR LED? You won't be able to see the LED even on max.

nseidle, 2018-03-28:
Closing due to inactivity. Please reopen if needed.

closed 2018-03-28 by nseidle