sparkfun/SparkFun_MAX3010x_Sensor_Library #7: MAX30105 + Arduino Pro Micro?

DonguiLLaume, 2017-12-06:
I’m trying to use this sensor together with an arduino pro micro. However, arduino pro micro has not neither SCL and SDA nor A4 and A5. Can I change something in the code to make the connexions to A2 and A3 instead? I am unable to find where to set this information. Thanks.

nseidle, 2018-02-15:
Yup. You can setup a second wire port on the [Pro Micro](https://www.sparkfun.com/products/12640) and then pass that stream into the library. I did something like this on my APRS entry. Checkout [this code](https://github.com/nseidle/APRS/blob/master/SensorCollator/Compass.ino#L155) to see how to init a 2nd port on the Pro Micro. Then call something like:

    myMax30105.begin(testI2C, 100000, 0x57);

This will init the MAX30105 using the testI2C i2c port, at 100kHz, using the I2C address of 0x57. See this [line](https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/blob/master/src/MAX30105.h#L50) to see what I'm talking about.

nseidle, 2018-02-15:
Also, see the **SERCOM – Configurable Serial Ports** section of the [SAMD21 hookup guide](https://learn.sparkfun.com/tutorials/samd21-minidev-breakout-hookup-guide/all).

closed 2018-02-15 by nseidle