sparkfun/SparkFun_MAX3010x_Sensor_Library #11: WeMos D1 resetting when I run Example8_SPO2

AbirKuundu, 2018-03-04:
Program Example8_SPO2 compiled successfully and got uploaded to Wemos D1 Mini.
Board I'm using MAX30102. Wire connection is as below:
WeMos D1 -> MAX30102
5V		       Vin
G		       Gnd
D1		       SCL
D2		       SDA
GND	       IRD (Without this my IR Led does light up) 

When I run the program, I receive the following error every 5 seconds. Seems like the board is resetting.  

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

Can you help? 
Note: When I run Example8_SPO2 on Arduino Nano, I can get SPO2 value from the board.

per1234, 2018-03-04:
Please test the following code and then let me know whether it works:
```cpp
/*
  Optical SP02 Detection (SPK Algorithm) using the MAX30105 Breakout
  By: Nathan Seidle @ SparkFun Electronics
  Date: October 19th, 2016
  https://github.com/sparkfun/MAX30105_Breakout

  This demo shows heart rate and SPO2 levels.

  It is best to attach the sensor to your finger using a rubber band or other tightening 
  device. Humans are generally bad at applying constant pressure to a thing. When you 
  press your finger against the sensor it varies enough to cause the blood in your 
  finger to flow differently which causes the sensor readings to go wonky.

  This example is based on MAXREFDES117 and RD117_LILYPAD.ino from Maxim. Their example
  was modified to work with the SparkFun MAX30105 library and to compile under Arduino 1.6.11
  Please see license file for more info.

  Hardware Connections (Breakoutboard to Arduino):
  -5V = 5V (3.3V is allowed)
  -GND = GND
  -SDA = A4 (or SDA)
  -SCL = A5 (or SCL)
  -INT = Not connected
 
  The MAX30105 Breakout can handle 5V or 3.3V I2C logic. We recommend powering the board with 5V
  but it will also run at 3.3V.
*/

#include <Wire.h>
#include "MAX30105.h"
#include "spo2_algorithm.h"

MAX30105 particleSensor;

#define MAX_BRIGHTNESS 255

#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega168__)
//Arduino Uno doesn't have enough SRAM to store 100 samples of IR led data and red led data in 32-bit format
//To solve this problem, 16-bit MSB of the sampled data will be truncated. Samples become 16-bit data.
uint16_t irBuffer[100]; //infrared LED sensor data
uint16_t redBuffer[100];  //red LED sensor data
#else
uint32_t irBuffer[100]; //infrared LED sensor data
uint32_t redBuffer[100];  //red LED sensor data
#endif

int32_t bufferLength; //data length
int32_t spo2; //SPO2 value
int8_t validSPO2; //indicator to show if the SPO2 calculation is valid
int32_t heartRate; //heart rate value
int8_t validHeartRate; //indicator to show if the heart rate calculation is valid

byte pulseLED = 11; //Must be on PWM pin
byte readLED = 13; //Blinks with each data read

void setup()
{
  Serial.begin(115200); // initialize serial communication at 115200 bits per second:

  pinMode(pulseLED, OUTPUT);
  pinMode(readLED, OUTPUT);

  // Initialize sensor
  if (!particleSensor.begin(Wire, I2C_SPEED_FAST)) //Use default I2C port, 400kHz speed
  {
    Serial.println(F("MAX30105 was not found. Please check wiring/power."));
    while (1);
  }

  Serial.println(F("Attach sensor to finger with rubber band. Press any key to start conversion"));
  while (Serial.available() == 0) ; //wait until user presses a key
  Serial.read();

  byte ledBrightness = 60; //Options: 0=Off to 255=50mA
  byte sampleAverage = 4; //Options: 1, 2, 4, 8, 16, 32
  byte ledMode = 2; //Options: 1 = Red only, 2 = Red + IR, 3 = Red + IR + Green
  byte sampleRate = 100; //Options: 50, 100, 200, 400, 800, 1000, 1600, 3200
  int pulseWidth = 411; //Options: 69, 118, 215, 411
  int adcRange = 4096; //Options: 2048, 4096, 8192, 16384

  particleSensor.setup(ledBrightness, sampleAverage, ledMode, sampleRate, pulseWidth, adcRange); //Configure sensor with these settings
    
  bufferLength = 100; //buffer length of 100 stores 4 seconds of samples running at 25sps

  //read the first 100 samples, and determine the signal range
  for (byte i = 0 ; i < bufferLength ; i++)
  {
    while (particleSensor.available() == false) //do we have new data?
      particleSensor.check(); //Check the sensor for new data

    redBuffer[i] = particleSensor.getRed();
    irBuffer[i] = particleSensor.getIR();
    particleSensor.nextSample(); //We're finished with this sample so move to next sample

    Serial.print(F("red="));
    Serial.print(redBuffer[i], DEC);
    Serial.print(F(", ir="));
    Serial.println(irBuffer[i], DEC);
  }

  //calculate heart rate and SpO2 after first 100 samples (first 4 seconds of samples)
  maxim_heart_rate_and_oxygen_saturation(irBuffer, bufferLength, redBuffer, &spo2, &validSPO2, &heartRate, &validHeartRate);
}

void loop()
{
  //Continuously taking samples from MAX30102.  Heart rate and SpO2 are calculated every 1 second

  //dumping the first 25 sets of samples in the memory and shift the last 75 sets of samples to the top
  for (byte i = 25; i < 100; i++)
  {
    redBuffer[i - 25] = redBuffer[i];
    irBuffer[i - 25] = irBuffer[i];
  }

  //take 25 sets of samples before calculating the heart rate.
  for (byte i = 75; i < 100; i++)
  {
    while (particleSensor.available() == false) //do we have new data?
      particleSensor.check(); //Check the sensor for new data

    digitalWrite(readLED, !digitalRead(readLED)); //Blink onboard LED with every data read

    redBuffer[i] = particleSensor.getRed();
    irBuffer[i] = particleSensor.getIR();
    particleSensor.nextSample(); //We're finished with this sample so move to next sample

    //send samples and calculation result to terminal program through UART
    Serial.print(F("red="));
    Serial.print(redBuffer[i], DEC);
    Serial.print(F(", ir="));
    Serial.print(irBuffer[i], DEC);

    Serial.print(F(", HR="));
    Serial.print(heartRate, DEC);

    Serial.print(F(", HRvalid="));
    Serial.print(validHeartRate, DEC);

    Serial.print(F(", SPO2="));
    Serial.print(spo2, DEC);

    Serial.print(F(", SPO2Valid="));
    Serial.println(validSPO2, DEC);
  }

  //After gathering 25 new samples recalculate HR and SP02
  maxim_heart_rate_and_oxygen_saturation(irBuffer, bufferLength, redBuffer, &spo2, &validSPO2, &heartRate, &validHeartRate);
}
```

All I did was remove the unnecessary endless `while()` loop. It's not needed because `loop()` is already an endless loop and it causes problems on ESP8266 because if `loop()` never returns then the watchdog timer doesn't get reset. The alternative solution would be to add a `yield()` or `delay(0)` statement but it's better to solve the problem by removing unnecessary code instead of adding confusing code.

AbirKuundu, 2018-03-04:
Now it gives a little different error code were the boot mode is different.

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

AthanasiosT, 2019-04-08:
The example8_SPO2 doesn't work on esp32 using max30102.
The compilation is ok but after loading the program on esp32 no leds light up on sensor and no data on serial monitor

miangelbarrio, 2020-01-23:
I have had this same problem with a NodeMCU board with ESP8266. This compiles the sketch well, but it fails and the board resets. I have solved it by removing the lines 
// pinMode (press LED, OUTPUT);
// pinMode (readLED, OUTPUT);
and all that made reference to them.

I do not understand why they are defined as output and later a digitalRead (readLED) is made
//digitalWrite(readLED, !digitalRead(readLED)); //Blink onboard LED with every data read

zlocimir, 2020-02-18:
@miangelbarrio Anything else?
I've done the same thing with my NodeMCU and it doesn't work...

janusf, 2020-07-11:
> The example8_SPO2 doesn't work on esp32 using max30102.
> The compilation is ok but after loading the program on esp32 no leds light up on sensor and no data on serial monitor

@AthanasiosT  did you finf a fix for this? I have the same problem using MKR1000 and MAX30102

theotherjenkutler, 2020-11-28:
I have the same problem with both ESP8266 and ESP32. The sensor is never found and the light never comes on, even if you remove the while(1) in the setup.

itthings, 2021-01-29:
Line 71 you need this:

while (Serial.available() == 0)//wait until user presses a key
  {
    yield();
  } 
  Serial.read();

HamzaHajeir, 2021-03-29:
> I have had this same problem with a NodeMCU board with ESP8266. This compiles the sketch well, but it fails and the board resets. I have solved it by removing the lines
> // pinMode (press LED, [...]

Ah

It's the forbidden pins on ESP8266 connected to flash. 

You touch them, They will hurt you!

ADITYA0012345, 2021-05-05:
> I have had this same problem with a NodeMCU board with ESP8266. This compiles the sketch well, but it fails and the board resets. I have solved it by removing the lines
> // pinMode (press LED, [...]

hey did you get the output after this?
Because I also made similar changes and still not getting anything. Although the light on the sensor is glowing and there's no error as well.