sparkfun/SparkFun_MAX3010x_Sensor_Library #17: HR betweem 150 to 250

talal44, 2018-08-10:
what am I missing here? what do I have to adjust to get the right numbers?
[image]

talal44, 2018-08-10:
Thanks for your prompt reply.
 I did everything as the guide explains, the spo2 looks great, however the hr is always high. 150 and above, that doesn't seem to be right. I am quite relaxed and alive.....

nseidle, 2018-08-11:
Let's check the basics.

What platform are you connected to? What voltage are you running at? Do you
have a link to the board you purchased?

talal44, 2018-08-11:
I have two boards, arduino uno and elegoo mega2560 r3. It doesn't make any difference whether I connect my sparkfun max30105 to  5v or 3.3v. I got almost same reading both ways.

nseidle, 2018-08-13:
And just to be clear, are you using the [SparkFun board](https://www.sparkfun.com/products/14045)? 

Looks like you're running example 8. Do you see different output when you run example 5_HeartRate? The IC is configured differently in Example 5.

talal44, 2018-08-14:
I am using the red Sparkfun max30105 board. I get better better sensible and logical hr numbers with example 5.

nseidle, 2018-08-14:
Ah, ok. Sounds like the hardware is functioning. Both examples depend on Maxim's algorithms. We can support the library but you'll need to talk to Maxim for support on the algorithms. 

A quick search turned up some other options. [This example](https://morf.lv/implementing-pulse-oximeter-using-max30100) might get you started down a better Sp02 detection.

talal44, 2018-08-19:
Thanks a lot for your guidance. However, I still didn't got it, am I the only one who has this problem with the HR numbers in example_8? even when I use my new Sparkfun esp8266 thing board I get the same results. When I opened this issue I was hoping that there is someone who had the same issue and found a solution somehow.

ThingEngineer, 2018-08-25:
I have run across the exact same problem. 
I'm running a MAX30102 on a DIYmall breakout board. http://a.co/d/d31zetS
On an ESP8266 (AI-Thinker) - (Adafruit Feather Huzzah equivalent)

All of the examples work as expected except the SPO2 example 8.
I get readings like this:
red=105481, ir=121830
red=105500, ir=121889
red=105502, ir=121942
red=105521, ir=122030, HR=166, HRvalid=1, SPO2=100, SPO2Valid=1
red=105529, ir=122088, HR=166, HRvalid=1, SPO2=100, SPO2Valid=1
red=105498, ir=121597, HR=166, HRvalid=1, SPO2=100, SPO2Valid=1
red=105229, ir=121368, HR=166, HRvalid=1, SPO2=100, SPO2Valid=1

The calculated values do fluctuate but they seem like complete garbage. 

My heart rate during the above measurement was between 50 and 60 bpm with a tight finger strap (velcro) and clean red/ir signals (seen if I switch to plotting them like this):
Serial.print(redBuffer[i], DEC);
Serial.print(F(","));
Serial.println(irBuffer[i] - 17000, DEC);

The - 17000 on the ir is to move it closer to the red so the plotter is zoomed in better. (I hate you Arduino serial plotter, but your convenient.)

I was hoping for a quick on chip solution to this so I could move on but looks like I'll have to dig in a bit more. I haven't dug into the yet but I'll share my results when I do.

pzink, 2018-11-17:
Having the same issue, same board as ThingEngineer. Anyone find another board/code example that works for SpO2?

anning865, 2018-12-25:
Same issue.
It is not about the board, it is about the code, or specifically the algorithm.
The code from Maxim can work sometimes, but it doesn't work well at the most time ,especially under the external light interference. I had checked the waveform of raw data and found it worse when I turned  on the table lamp next to me. So you can use some black item to cover the sensor board and try it again.   But I still think the code from Maxim is too sensitive to the interference and we need to improved it.

soluna1, 2020-05-17:
Hi, couple of years later still finding the same issue with HR with using code from example 8, based on spo2_algorithm.h library. But, I get reasonable good data with example 5, based on #include "MAX30105.h" lib. So, has anyone solved this questions. Is it possible to combine HR calculated from example 5 and spo2 from example 8? BTW both seems to work fin in MAX30100.
I hope there is still someone interested in this. 

thanks

caferrick, 2020-05-17:
I am also having the same issue. Heart Rate is just all over the place the Oxygen reading seems good. When i combine 5 and 8 the heart rate is still all over the place. Soluna1 I will keep report back with any additional information.

lfssp, 2020-06-15:
I am also struggling with this issue! Example 5 works like a charm for HR. Example 8 shows SPO correctly, but HR is completely messed up. I've been trying to combine them, but apparently the issue is when you call setup() with different values. Setup() conf for example 5 doesn't work with spo and setup parameters from example 8 doesn't get correct HR.

caferrick, 2020-06-15:
I have a new algorithm that seems to work fairly well. I will reply to this
email in a little bit.

lfssp, 2020-06-15:
That's a good news @caferrick . Waiting to see your progress.

caferrick, 2020-06-16:
I did not strip anything out so please use what you like. The program send
the SPO2 and HBM via bluetooth. The device that I am using is an Arduino
feather.

/************************************
 *
 * This includes are bluetooth
 *
 ***********************************/
#include <Arduino.h>
#include <SPI.h>
#include "Adafruit_BluefruitLE_SPI.h"
#include "Adafruit_BluefruitLE_UART.h"

#include "BluefruitConfig.h"

// BlueTooth
#if SOFTWARE_SERIAL_AVAILABLE
  #include <SoftwareSerial.h>
#endif

Adafruit_BluefruitLE_SPI ble(BLUEFRUIT_SPI_CS, BLUEFRUIT_SPI_IRQ,
BLUEFRUIT_SPI_RST);

#include <Wire.h>
#include "MAX30105.h" //sparkfun MAX3010X library
MAX30105 particleSensor;

//#define MAX30105 //if you have Sparkfun's MAX30105 breakout board , try
#define MAX30105

/* BlueTooth service information */

int32_t ServiceId;
int32_t hrmMeasureCharId;
int32_t hrmLocationCharId;
int32_t spoMeasureCharId;

/****************** heart rate *********************/
static double fbpmrate = 0.95; // low pass filter coefficient for HRM in bpm
  static uint32_t crosstime = 0; //falling edge , zero crossing time in msec
  static uint32_t crosstime_prev = 0;//previous falling edge , zero
crossing time in msec
  static double bpm = 40.0;
  static double ebpm = 40.0;
  static double eir = 0.0; //estimated lowpass filtered IR signal to find
falling edge without notch
  static double firrate = 0.85; //IR filter coefficient to remove notch ,
should be smaller than frate
  static double eir_prev = 0.0;

#define TIMETOBOOT 3000 // wait for this time(msec) to output SpO2
#define SCALE 88.0 //adjust to display heart beat and SpO2 in the same scale
#define SAMPLING 5 //if you want to see heart beat more precisely , set
SAMPLING to 1
#define FINGER_ON 30000 // if red signal is lower than this , it indicates
your finger is not on the sensor
#define MINIMUM_SPO2 80.0

#define FINGER_ON 50000 // if ir signal is lower than this , it indicates
your finger is not on the sensor
#define LED_PERIOD 100 // light up LED for this period in msec when zero
crossing is found for filtered IR signal
#define MAX_BPS 180
#define MIN_BPS 45

// A small helper
void error(const __FlashStringHelper*err) {
  Serial.println(err);
  while (1);
[... 225 lines cut ...]
      //Serial.println(CTdiff);

    //if ( ((crosstime - crosstime_prev ) > (60 * 1000 / MAX_BPS)) &&
((crosstime - crosstime_prev ) < (60 * 1000 / MIN_BPS)) ) {

    if ( ( CTdiff > 333 ) && ( CTdiff < 1333 ) ) {
      bpm = 60.0 * 1000.0 / (double)(crosstime - crosstime_prev) ; //get bpm
        // Serial.println("crossed");
      ebpm = ebpm * fbpmrate + (1.0 - fbpmrate) * bpm;//estimated bpm by
low pass filtered
#ifdef LED_SOUND_INDICATOR
      if (aveir > FINGER_ON) {
        digitalWrite(LEDPORT, HIGH);
        tone(BLIPSOUND-(100.0-ESpO2)*10.0);//when SpO2=80% BLIPSOUND drops
200Hz to indicate anormaly
      }
#endif
    } else {
      //Serial.println("faild to find falling edge");
    }
    crosstime_prev = crosstime;
  }
   eir_prev = eir;
   return (ebpm);
}

soluna1, 2020-06-16:
Hi @caferrick , thank you for the updated code!! I'll let you know asap how it works. Sure it's great!

linkindrew, 2020-06-16:
@caferrick Nice job up there! (using MAX30102 + ESP8266) I Eliminated the Bluetooth relevant code, I can see drastic improvement in HR but SpO2 is pretty unstable in for me here (spanning 60-110 randomly)

linkindrew, 2020-06-16:
(using MAX30102 + ESP8266) Example_08 for HR+SpO2 is giving my random values, not even being close to accurate, with SpO2 being -999, invalid for the most part

caferrick, 2020-06-16:
I noticed with the sensor that I am using my finger has to be fairly stable
on the device. Also the 20 seconds or the numbers are all over the place

linkindrew, 2020-06-16:
@caferrick can you share any supporting research/study based on which, you have come up with your algortihm? I'd like to tweak `FSpO2` or something there to make SpO2 consistent, a top threshold of 100 is also required.

caferrick, 2020-06-16:
I found the original Algorithm on the Web and modified it slightly based on
the MAX3012 data sheet. I will try to tweak the code a little more and if
you have any luck please let me know.

linkindrew, 2020-06-16:
@caferrick Can you share the literature/link? That'd help me to better understand various other aspects/factors and i may be able to effectively tweak the code.

caferrick, 2020-06-16:
here is just on site

https://pdfserv.maximintegrated.com/en/an/AN6409.pdf   You can search for
spec sheets for MAX30102

linkindrew, 2020-06-16:
I see, i have the specs sheet, was trying various MAX30101 algorithms to run on MAX30102 by updating registers mapping, didn't get any success yet.

linkindrew, 2020-06-16:
@caferrick which sensor are you using? is it Maxim's 30102 or some other vendor?

caferrick, 2020-06-17:
30102 5 connector pins on one side the other just had the LED and two
mounting holes

caferrick, 2020-06-17:
this is the unit that i am using
 HiLetgo MAX30102 Low Power Heart Rate Click Sensor Breakout Board Module
for Arduino Pulse Oximetry Solution SpO2 Replace MAX30100
<https://www.amazon.com/gp/product/B07QC67KMQ/ref=ppx_yo_dt_b_asin_title_o09_s01?ie=UTF8&psc=1>

bbogdanmircea, 2020-06-17:
Hello, I just want to confirm the same thing about the examples for the SparkFun library.
The heart Rate Plotter and Normal example seems to work nice.
But the SPO2 example seems to show very unstable values.

I tried to run different drivers with other algorithms, that I found on instructables and the web, but most algorithms use the interrupt line from the sensor.
I found 2 sources that have a very good documentation for the algorithms, I will post them here.
I am using the driver on a MAX30102 board that I bought from eBay, the pullups for the I2C signals where wrongly connected to 1.8V so I had to cut the track and solder them to 3.3V.
The code is running on a TTGO ESP32 board that has also an OLED display, I hope that I can display the graphic for the heart rate on the display like in the Plotter and make some nice menus for other options, but first I need some reliable algorithms.

bbogdanmircea, 2020-06-18:
Hello,

I took the code posted by caferrick and made a sketch out of it.
I made some modifications, added some defines for the BT board and also for various debugging on Serial.
The signal for the Heart Beat looks ok from what I can see, but the SPO2 calculation seems to be still wrong.
This is running on an ESP32 board, but it should also work on an Arduino without problems.

[caferrick_max30100.zip](https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/files/4799184/caferrick_max30100.zip)

caferrick, 2020-06-18:
I wonder why the SPO2 is an issue on my Arduino Feather. My SPO2 seems
good. You have wait about 30 seconds for the readings to normalize.

linkindrew, 2020-06-19:
I just discovered that chinese replica breakout boards has swapped IR and LED registers! that caused me all the trouble (using MH-ET Live Max30102). Here is [my implementation of pulse oximeter](https://github.com/linkindrew/pulseoximeter_max30102), a port of [MAX30102_by_RF]https://github.com/aromring/MAX30102_by_RF) for NodeMCU that works perfectly for Original Maxim or chinese replica sensors.

kiranshashiny, 2020-08-26:
I am having the same problem - the number is all over the place 150 to 200, and it cycles thru.
Hope the developers upgrade or update the code to newest !

ROdoka, 2021-03-29:
I don't think anyone develops at all :o
I just started a project with MAX30102 and try to figure out how to use it (and we want to use it on a horse, don't ask ;) )
IOW: I have the same problems like the others here, HR is somewhere between 125 and 250 on an Arduino Uno (since ESP32 doesn't work)

Swadhin21, 2021-04-17:
I am also facing the same problem I.e. the heart rate values are going above 150BPM but the SpO2 value seems to be accurate in the example 8 of spark fun library. So, please anyone can help me if anyone have solved the problem

caferrick, 2021-04-17:
Did you try the code I listed a few months ago?  I am using an Arduino
Feather U32 and I am not having the issue.

ROdoka, 2021-04-19:
actually, I think I didn't try it, I had to find a quick working solution because of deadline reasons and got the sparkfun sensor hub PCB with MAX30101 and MAX32664, this works fine with its lib on ESP32.

after I got the ESP back, I will try your code above.

perohlin, 2021-05-04:
hello
some times with tuning led brightnes and lowest in 0x10 to 0x20 module work fine but not always. i think this module is very sensetive to distance between finger and chip and config parameters

anuragmehra24, 2021-05-12:
Hi
i am also getting issues in accuracy of heart rate. I tried the code shared by caferrick but heart rate is not coming accurate. Even in just sitting positing it is going above 100 using MAX30102 sensor and when i measure with mobile apps which are using mobile flash and mobile camera, the heart rate is coming in normal range.

caferrick, 2021-05-12:
I am also using MAX30102 with Arduino Feather I am not having the issue
that you are seeing. Do you try the suggestion from perohlin. These sensors
can be a bit touchy

capitalfuse, 2021-05-18:
> I just discovered that chinese replica breakout boards has swapped IR and LED registers! that caused me all the trouble (using MH-ET Live Max30102). Here is [my implementation of pulse [...]

I bought Max30102 module at Aliexpress. I tried Spo2 sample code on ESP32, but it doesn't seem to work correctly. Also the presence sample code doesn't (Red LED is disabled to be 0 power in this code). 
So, I changed the followings;

> LED1 < --- > LED2
> sense.red < --- > sense.IR

**max30105.cpp**
```
.....
.....
void MAX30105::setPulseAmplitudeRed(uint8_t amplitude) {
  writeRegister8(_i2caddr, MAX30105_LED2_PULSEAMP, amplitude);
}

void MAX30105::setPulseAmplitudeIR(uint8_t amplitude) {
  writeRegister8(_i2caddr, MAX30105_LED1_PULSEAMP, amplitude);
}
.....
.....
.....
while (toGet > 0)
      {
        sense.head++; //Advance the head of the storage struct
        sense.head %= STORAGE_SIZE; //Wrap condition

        byte temp[sizeof(uint32_t)]; //Array of 4 bytes that we will convert into long
        uint32_t tempLong;

        //Burst read three bytes - RED
        temp[3] = 0;
        temp[2] = _i2cPort->read();
        temp[1] = _i2cPort->read();
        temp[0] = _i2cPort->read();

        //Convert array to long
        memcpy(&tempLong, temp, sizeof(tempLong));
		
		tempLong &= 0x3FFFF; //Zero out all but 18 bits

        sense.IR[sense.head] = tempLong; //Store this reading into the sense array

        if (activeLEDs > 1)
        {
          //Burst read three more bytes - IR
          temp[3] = 0;
          temp[2] = _i2cPort->read();
          temp[1] = _i2cPort->read();
          temp[0] = _i2cPort->read();

          //Convert array to long
          memcpy(&tempLong, temp, sizeof(tempLong));

		  tempLong &= 0x3FFFF; //Zero out all but 18 bits
          
		  sense.red[sense.head] = tempLong;
        }

``` 
Got good results for HR and Spo2 outputs. This is Right???

caferrick, 2021-05-18:
Nice!

AhmedAymanNegmm, 2021-05-21:
> what am I missing here? what do I have to adjust to get the right numbers?
> [image]

@talal44  How did you get correct SpO2 values? I am using example 8 and getting very wrong numbers. please help

capitalfuse, 2021-05-26:
Get more accurate value after changing to "byte sampleAverage = 1;" in example 8.
Its HR value almost matches with HR value in example 5.
But I don't understand why that changed. Anyone explain about it?
(I checked it on ESP32+MAX30102)

denisandroid, 2022-01-17:
**MH-ET LIVE, rev: 3 part: 15, esp8266**

I confirm that the Chinese MAX30102 sensor assembled on a black scarf has the correct pinout and 3.3 volt pull-ups, but the red LED and infrared are confused with each other from the factory.

Without altering the libraries to a different order of receiving I2C data, the sensor refuses to give adequate values at all, it's not about sampling or configuration, it's about this feature. 

<details>
  <summary>Code of another adapted library (MAX30102_by_RF)</summary>
    
    bool maxim_max30102_read_fifo(uint32_t *pun_red_led, uint32_t *pun_ir_led) {
      uint32_t un_temp;
      
      *pun_ir_led=0;
      *pun_red_led=0;
      
      // read 1 byte from registers (values are discarded)
      maxim_max30102_eventread(REG_INTR_STATUS_1);
      maxim_max30102_eventread(REG_INTR_STATUS_2);
    
      yield();
      Wire.beginTransmission(I2C_WRITE_ADDR);
      Wire.write(REG_FIFO_DATA);
      Wire.endTransmission();
    
      delayMicroseconds(1);
      
      yield();
      Wire.beginTransmission(I2C_READ_ADDR);
      Wire.requestFrom(I2C_READ_ADDR,6);
    
      #if defined(IS_EXISTS_SWAP_IR_AND_RED_LED) 
        un_temp = Wire.read();
        un_temp <<= 16;
        *pun_ir_led += un_temp;  // IR!
        un_temp = Wire.read();
        un_temp <<= 8;
        *pun_ir_led += un_temp;
        un_temp = Wire.read();
        *pun_ir_led += un_temp;
        
        un_temp = Wire.read();
        un_temp <<= 16;
        *pun_red_led += un_temp; // RED
        un_temp = Wire.read();
        un_temp <<= 8;
        *pun_red_led += un_temp;
        un_temp = Wire.read();
        *pun_red_led += un_temp;
      #else 
        un_temp=Wire.read();
        un_temp<<=16;
        *pun_red_led += un_temp; // RED!
        un_temp=Wire.read();
        un_temp<<=8;
        *pun_red_led += un_temp;
        un_temp=Wire.read();
        *pun_red_led += un_temp;
        
        un_temp=Wire.read();
        un_temp<<=16;
        *pun_ir_led += un_temp; // IR!
        un_temp=Wire.read();
        un_temp<<=8;
        *pun_ir_led += un_temp;
        un_temp = Wire.read();
        *pun_ir_led += un_temp;
      #endif
      
      
      Wire.endTransmission();
      
      *pun_red_led &= 0x03FFFF;  //Mask MSB [23:18]
      *pun_ir_led &= 0x03FFFF;  //Mask MSB [23:18]
      
      return true;
    }

</details>

matheuszap, 2022-02-02:
Has anyone managed to run max30105 with esp8266? I'm trying but I don't have any results. In the compilation everything happens normally, but when I run it no LED lights up.

VitoDrofenik, 2022-02-28:
@matheuszap I am not using the 30105 but the 30102 and on a custom board with a STM32L processor but I think the setup should be quite the same. Did you set the 0x0C, 0x0D and 0x0E registers? Also check that you are not setting bit 7 to 1 in the 0x09 register.

mariagambelli5, 2022-05-05:
@caferrick With my group we trying to use your code for sensor max30102 with an Arduino UNO because we really appreciate your code but we have some issues because we aren't expert in this field. Have you any advice for using this code for an Arduino UNO in order to obtain reasonable values of HR and SpO2? Thanks for helping us!

VitoDrofenik, 2022-05-05:
What seems to be your problem?
Maybe include some raw red and ir readings (at least 100).

mariagambelli5, 2022-05-05:
Thanks for answering us. Did you use the cafericck code with a different type of Arduino? We have already tried your idea but without any success. We also tried to join the example 8 and example 5 of sparkfun library for max30102 sensor but we don't manage to obtain a right code. Did you try to do that?

VitoDrofenik, 2022-05-05:
> I just discovered that chinese replica breakout boards has swapped IR and LED registers! that caused me all the trouble (using MH-ET Live Max30102). Here is [my implementation of pulse [...]

No. I used MAX30102 on a custom STM32 board. I did some tweaks but mostly the code by RF should be what you need if you go through it and modify it for your board. If you need, you can publish your code and what you are getting. Either way I will have a look at it tomorrow. What helped me get the correct settings was saving the last x seconds of raw IR and RED readings and then I put them in a graphing tool to see if the signal is clear or not (excel or similar can also work)

mariagambelli5, 2022-05-06:
Hi! Sorry for the delay, we try to delete from the @caferrick code reported above all the parts that refer to the bluetooth connection because we only need the values of heart rate and SpO2 using the max30102 and an Arduino UNO in the serial monitor. With the following code we manage to obtain reasonable values of SpO2 but we always see the heart rate equals to 1 but we don't know why. Have you any ideas? Thanks for helping us!

#include <Wire.h>
#include "MAX30105.h" //sparkfun MAX3010X library
MAX30105 particleSensor;

//#define MAX30105 //if you have Sparkfun's MAX30105 breakout board , try
#define MAX30105

/****************** heart rate *********************/
static double fbpmrate = 0.95; // low pass filter coefficient for HRM in bpm
  static uint32_t crosstime = 0; //falling edge , zero crossing time in msec
  static uint32_t crosstime_prev = 0;//previous falling edge , zero crossing time in msec
  static double bpm = 40.0;
  static double ebpm = 40.0;
  static double eir = 0.0; //estimated lowpass filtered IR signal to find falling edge without notch
  static double firrate = 0.85; //IR filter coefficient to remove notch , should be smaller than frate
  static double eir_prev = 0.0;

#define TIMETOBOOT 3000 // wait for this time(msec) to output SpO2
#define SCALE 88.0 //adjust to display heart beat and SpO2 in the same scale
#define SAMPLING 5 //if you want to see heart beat more precisely , set SAMPLING to 1
#define FINGER_ON 30000 // if red signal is lower than this , it indicates your finger is not on the sensor
#define MINIMUM_SPO2 80.0

#define FINGER_ON 50000 // if ir signal is lower than this , it indicates your finger is not on the sensor
#define LED_PERIOD 100 // light up LED for this period in msec when zero crossing is found for filtered IR signal
#define MAX_BPS 18
#define MIN_BPS 45

// A small helper
void error(const __FlashStringHelper*err) {
  Serial.println(err);
  while (1);
}

#define USEFIFO
void setup()
{

  boolean success;
  Serial.begin(115200);

  /* Perform a factory reset to make sure everything is in a known state */
  Serial.println(F("Performing a factory reset: "));
 
  /* Change the device name to make it easier to find */
  Serial.println(F("Setting device name to 'Bluefruit HRM': "));
  Serial.println();

  // Initialize sensor
  if (!particleSensor.begin(Wire, I2C_SPEED_FAST)) //Use default I2C port, 400kHz speed
  {
    Serial.println("MAX30102 was not found. Please check wiring/power/solder jumper at MH-ET LIVE MAX30102 board. ");
    while (1);
  }

  //Setup MAX30102
  byte ledBrightness =70; //Options: 0=Off to 255=50mA
  byte sampleAverage = 2; //Options: 1, 2, 4, 8, 16, 32
  byte ledMode = 2; //Options: 1 = Red only, 2 = Red + IR, 3 = Red + IR + Green //Options: 1 = IR only, 2 = Red + IR on MH-ET LIVE MAX30102 board
  int sampleRate = 400; //Options: 50, 100, 200, 400, 800, 1000, 1600, 3200
  int pulseWidth = 411; //Options: 69, 118, 215, 411
  int adcRange = 4096; //Options: 2048, 4096, 8192, 16384
  // Set up the wanted parameters
  particleSensor.setup(ledBrightness, sampleAverage, ledMode, sampleRate, pulseWidth, adcRange); //Configure sensor with these settings

}

double avered = 0; double aveir = 0;
[... 99 lines cut ...]
      //Serial.println(CTdiff);

      
    if ( ((crosstime - crosstime_prev ) > (60 * 1000 / MAX_BPS)) && //questo if era commentato solo la riga sopra 
((crosstime - crosstime_prev ) < (60 * 1000 / MIN_BPS)) ) {

    if ( ( CTdiff > 333 ) && ( CTdiff < 1333 ) ) {
      bpm = 60.0 * 1000.0 / (double)(crosstime - crosstime_prev) ; //get bpm
        // Serial.println("crossed");
      ebpm = ebpm * fbpmrate + (1.0 - fbpmrate) * bpm;//estimated bpm by low pass filtered provo a mettere la maiuscola
/*#ifdef LED_SOUND_INDICATOR 
      if (aveir > FINGER_ON) {
        digitalWrite(LEDPORT, HIGH);
        tone(BLIPSOUND-(100.0-ESpO2)*10.0);//when SpO2=80% BLIPSOUND drops 200Hz to indicate anormaly
      }
#endif*/  //dovrebbe servire solo a dare un avviso 
    } else {
      Serial.println("faild to find falling edge");
    }
    crosstime_prev = crosstime;
  }
   eir_prev = eir;
   return (ebpm);
}
}

josdiaz-tech, 2022-11-22:
Is there a solution to this issue? I want HR data below 120 and accurate SpO2 data. I don't care much about stability in HR , I just want an HR that is not crazy high

caferrick, 2022-11-23:
I never had the problem that you are seeing. Are you using the sample code
that I provided ?

JarenElectron, 2023-10-24:
SO... for whats its worth another year later. I have been working with the MAX30102 for weeks trying to understand why Maxim's/Sparkfuns implementation of Example 8 does not work at all for heartrate. When I started looking at raw data in excel I noticed that the samples were 80ms apart. That is only 12.5ish s/s. Half of what the algorithm was built around?

Anyway I modified *pn_heart_rate =(int32_t)( (FreqS*22.3)/ (n_peak_interval_sum) ); in the file spo2_algorithm. Its just a ratio. So I modified the left hand number until I was able to match my commercial solution. There is a little bit of jitter (courtesy of movement artifacts) but it measures my HR dead on. I get up - do the chicken dance - and see the number match on both the Serial print out and my commercial device.

They fall at the same rate to - although the Arduino version does lag. SPO2 reports accurately as well - hyperventilating and such to test this.

Why is this relevant? And what do I need to fix to reach 25sps (the stated design speed)?

yh00git, 2023-10-29:
Big thanks to @caferrick  @capitalfuse  <3
==================================================================================
Arduino MEGA user here.
I'm using MAX30102, 
Only connection of V in, SDA, SCL and Ground needed.
Conclusion : Result is very close to an actual pulse oximeter.  

After readings all the comments and trying all possible codes. 
This is what i discovered so far.

Step (1) 
I have the same problem/ same china replica of MAX30102 with @capitalfuse . Thus, I followed his/her instruction toto change some of the code in MAX30105.cpp

The .cpp file can found in:
C:\Users\User\Documents\Arduino\libraries\SparkFun_MAX3010x_Pulse_and_Proximity_Sensor_Library\src

```
> The changes are:
> LED1 < --- > LED2
> sense.red < --- > sense.IR
```
Result after modification: the result of HR and SpO2 of example8 seems more logic now.

The segment of code that need modification in MAX30105.cpp were ( I will mention the codes here in case u guys missed it): 

**MAX30105.cpp**
```
.....
.....
void MAX30105::setPulseAmplitudeRed(uint8_t amplitude) {
  writeRegister8(_i2caddr, MAX30105_LED2_PULSEAMP, amplitude);
}

void MAX30105::setPulseAmplitudeIR(uint8_t amplitude) {
  writeRegister8(_i2caddr, MAX30105_LED1_PULSEAMP, amplitude);
}
.....
.....
.....
while (toGet > 0)
      {
        sense.head++; //Advance the head of the storage struct
        sense.head %= STORAGE_SIZE; //Wrap condition

        byte temp[sizeof(uint32_t)]; //Array of 4 bytes that we will convert into long
        uint32_t tempLong;

        //Burst read three bytes - RED
        temp[3] = 0;
        temp[2] = _i2cPort->read();
        temp[1] = _i2cPort->read();
        temp[0] = _i2cPort->read();

        //Convert array to long
        memcpy(&tempLong, temp, sizeof(tempLong));
		
		tempLong &= 0x3FFFF; //Zero out all but 18 bits

        sense.IR[sense.head] = tempLong; //Store this reading into the sense array

        if (activeLEDs > 1)
        {
          //Burst read three more bytes - IR
          temp[3] = 0;
          temp[2] = _i2cPort->read();
          temp[1] = _i2cPort->read();
          temp[0] = _i2cPort->read();

          //Convert array to long
          memcpy(&tempLong, temp, sizeof(tempLong));
[... 228 lines cut ...]
    #endif

    //if ( ((crosstime - crosstime_prev ) > (60 * 1000 / MAX_BPS)) &&((crosstime - crosstime_prev ) < (60 * 1000 / MIN_BPS)) ) {

    if ( ( CTdiff > 333 ) && ( CTdiff < 1333 ) ) {
      bpm = 60.0 * 1000.0 / (double)(crosstime - crosstime_prev) ; //get bpm
      // Serial.println("crossed");
      ebpm = ebpm * fbpmrate + (1.0 - fbpmrate) * bpm; //estimated bpm by low pass filtered
      
  #ifdef LED_SOUND_INDICATOR
      if (aveir > FINGER_ON) {
        digitalWrite(LEDPORT, HIGH);
        tone(BLIPSOUND-(100.0-ESpO2)*10.0);//when SpO2=80% BLIPSOUND drops 200Hz to indicate anormaly
      }
  #endif
    } else {
      //Serial.println("faild to find falling edge");
    }
    crosstime_prev = crosstime;
  }
   eir_prev = eir;
   return (ebpm);
}

```

rayhannarindran, 2023-11-26:
> SO... for whats its worth another year later. I have been working with the MAX30102 for weeks trying to understand why Maxim's/Sparkfuns implementation of Example 8 does not work at all for heartrate. When I started looking at raw data in excel I noticed that the samples were 80ms apart. That is only 12.5ish s/s. Half of what the algorithm was built around?
> 
> So with that in mind I started to look for numbers related to sample frequency in some way to change - Anyway I modified _pn_heart_rate =(int32_t)( (FreqS_22.3)/ (n_peak_interval_sum) ); in the file spo2_algorithm. Its just a ratio. So I modified the left hand number until I was able to match my commercial solution. There is a little bit of jitter (courtesy of movement artifacts) but it measures my HR dead on. I get up - do the chicken dance - and see the number match on both the Serial print out and my commercial device.
> 
> They fall at the same rate to - although the Arduino version does lag. SPO2 reports accurately as well - hyperventilating and such to test this.
> 
> Why is this relevant? And what do I need to fix to reach 25sps (the stated design speed)?

@JarenElectron can you show the sketch/code on the files that you've changed, did you change the .h or .cpp file of the lib? what specifically did you change on those libs, thank you

JarenElectron, 2023-11-26:
@rayhannarindran 

spo2_algorithm.cpp

Line 138. Changed from OG to before mentioned.

Again, my logic being, that when I started trying to collect a bunch of samples to look at in excel and visualize the math - I noticed that the s/s we're about half of the expected 25 s/s. So I started looking for things in the code that deal with sample rate. While strapping a commercial transmissive pulse oximeter to my hand to verify the results...

I'd love it if someone can explain my solution and an the underlying problem...

rayhannarindran, 2023-11-27:
> @rayhannarindran
>
> spo2_algorithm.cpp
>
> Line 138. Changed from OG to before mentioned.
>
> Again, my logic being, that when I started trying to collect a bunch of samples to look at in excel and [...]

@JarenElectron yep it seemed to worked, i compared it to Example5, and the output is fairly similar.

FaithCyril-star, 2024-03-25:
my work around was to measure the sp02 for the first few minutes and the latter part of the time, measure the heart rate . I created separate functions for this by repurposing the code in the example sketches provided by SparkFun. here is the link to it but it sits among other code, https://github.com/FaithCyril-star/RPMS/blob/main/BAN_firmware/firmware/firmware.ino

[1 more reply]