sparkfun/SparkFun_MAX3010x_Sensor_Library #20: Example8_SPO2 doesn't work on Espressif ESP32

AthanasiosT, 2019-04-09:
The example8_SPO2 doesn't work on esp32 using max30102.
The compilation is ok but after loading the program on esp32 no leds light up on sensor and no data on serial monitor

AthanasiosT, 2019-04-09:
After pressing enter in serial monitor system start but values are abnormal. Should I use pullup resistors?

AthanasiosT, 2019-04-15:
I used pullup resistors, I used 3.3V and 5V but values of SpO2 and Heartrate are abnormal.

AhmedAymanNegmm, 2021-05-21:
Hi. Did you find any solution for getting wrong SpO2 values? I am also using example 8 and getting very far values (range from 10 to 100, and sometimes -999)

capitalfuse, 2021-05-25:
Please refer the following;
https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/issues/17#issuecomment-843191546