sparkfun/SparkFun_MAX3010x_Sensor_Library #22: This library should work with other MAX3010x sensors

Kabron287, 2019-06-06:
This library should work with other MAX3010x sensors including the MAX30102, MAX30101, and MAX30100.
How could it be if MAX30105 ID (declared in MAX30105.cpp in string #136) = 0x15, 
while e.g. MAX30100 ID = 0x11 ?

Kabron287, 2019-06-06:
And I, in general, do not understand why this chip(MAX30105) is mentioned here , which is an air pollution sensor.

nseidle, 2019-06-25:
You're sort of correct. Currently the library is hardcoded on [this line](https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/blob/master/src/MAX30105.cpp#L153) to check for the correct ID from the MAX30105. I haven't played with the MAX30100 but I have with the 30102 and this library works great (minus the fact that the 30102 doesn't have a green LED). This library could be extended to support the other ICs. I'm not sure what the 30100 has vs doesn't have compared to the 30105. If you'd like to dig in with me and build it out I'd be happy to.

The MAX30105 was originally sold as a smoke detection sensor but Maxim seems to have never developed a reliable algorithm so they've dropped that aspect of the marketing around the sensor. But the 30105 still works most excellently as a heart rate and pulse ox sensor.

FelipeMdeO, 2019-06-25:
This library no will work with MAX30100 because memory map of it is different.
Looking for:
Max30100 - >
[image]

To MAX30102, you have:
[image]

Max30102 have same memory map of Max30105, so you can use both with same* base library.

FelipeMdeO, 2019-07-23:
I found other point that we need atention.

In MAX30102 we no have function of Proximity mode. It was removed in last release:
[image]

particleincell, 2020-04-13:
Re Nathan's comment that Maxim never developed a reliable smoke detection algorithm, this is quite disappointing since that is what I actually got the board for. Have you had any luck actually detecting particulate levels with the sensor and if so, any suggestions how to interpret the data? I left the sensor logging for some 24 hours and while it seems the IR signal started trending downward overnight (with nobody in the office), there is just so much noise.

PeterAJansen, 2020-04-13:
@particleincell I had a go at trying to align the output of the MAX30105 with several other particulate matter sensors a few years ago, and couldn't figure out something reliable in the small amount of effort that I put into it -- especially without some other amount of external optics.  I'm not sure how they were doing it internally, with the externally licensed algorithm I think I remember them saying you could purchase. 

Here's more info on my attempt.  FYI -- I just use a proper particle sensor now (e.g. plantronics), though they are obviously much larger. 

https://hackaday.io/project/18518-iteration-8/log/55721-a-first-attempt-at-figuring-out-the-max30105-air-particle-sensor

kentpaul, 2020-05-02:
can someone help me with this issue. im uisng nodemcu esp8266 and max30102. and i get this wdt reset. im a beginner in this 

`Soft WDT reset

stack>>>

ctx: cont
sp: 3ffffde0 end: 3fffffc0 offset: 01b0
3fffff90: feefeffe feefeffe feefeffe 3ffee650
3fffffa0: 3fffdad0 00000000 3ffee610 402025cc
3fffffb0: feefeffe feefeffe 3ffe84ec 40100ea1
<<<stack<<<

ets Jan 8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld
Initializing MAX30100..FAILED: I2C error`

Tomisdebom, 2021-05-06:
Has anyone gotten this to work for the MAX30101?