sparkfun/SparkFun_MAX3010x_Sensor_Library #24: Is it possible to send the data over Wi-Fi (ESP8266)?

Frunner20, 2020-03-06:
I have no idea on how to make something that will send spo2 and heart rate to a remote server (w/ an interval in 1 sec). Can anyone please help? Some useful code would be highly appreciated.

mike2nl, 2020-04-24:
@Frunner20 
just found back my sensor in my box because i had have 4 hart attacks and will work now on a human sensor array. The board will be the arduino mega with esp8266 onboard. The mega because i will use lots of different sensors.

The boad is from RobotDyn. They hava a smaller one too. The UNO+WIFI. Menas the arduino uno and esp8266 onboard. I have them to for other projects.

First i will play with the sensor then work further. There is the Tasmota Project on github and i'm one of the admins in this. I write drivers and test a lot. There is a driver named Tasmota-Slave. That one is talking via rs232/serial from the esp to slave(onu, nano,....) and also backwards. Commands and reqeusts from both sides.

At the moment there is no value visible on the Tasmota WEB UI, only via mqtt, but....

I will use the base driver so as described in the Tasmota WIki and then make a complete new software with wifi, web ui, ftp or stream data, mqtt is a second thogth but away from Tasmota because i don't need the rest.

I will use the following sensors:
- Hart rate and Oxymeter
- UV sensor the veml6070 (i wrote a very good calibrated driver for it)
- Temperature, Humidity, AIr pressure and AQ = menas air quality (BME680)
- Multi Channel GAS sensor from Seeedstudio in the Grove series
- possible some others

SO when i have something i can inform you or we work togehter.
I ahve my own Discod Sensor Channel, haha, but i'm the only one right now.

When you are interested PM me on The tasmota discord channel under 'mike2nl'
discord: https://discord.gg/Ks2Kzd4

kentpaul, 2020-05-03:
hi sir i have this issue on my max30100 using nodemcu esp8266 maybe you can help me. im having an issue of soft wdt reset in my ouput.