sparkfun/SparkFun_MAX3010x_Sensor_Library #25: Wrong activation of R and IR in MAX230102?

soluna1, 2020-05-09:
Hi, I'm playing with a MAX30102 sensor using this library on Arduino Nano board. Trying to configure the code I came across with some guy who used this sensor in a research project. And he comments that if the library is used with MAX30102 the function getRed() actually gets the IR value, and viceversa. Could someone confirm if it's necessary to modify this in the code on the examples. As for comprobación, if ledMode = 1 no light shines, when supposedly should be a Red light, so it seems that Red is activates when IR is called, and IR when Red in called. 

Thanks,

ananevilya, 2020-10-02:
Hi @soluna1, 
It seems that MAX30101 and MAX30102 are the same. I have checked their datasheets. All registers are the same. MAX30101 only have extra two green led registers.
But I still can't figure out why MAX30102 can't work with this library.

nielsole, 2020-12-31:
Just for reference supporting what ananevilya said:
https://datasheets.maximintegrated.com/en/ds/MAX30102.pdf Page 18 Table 4
https://datasheets.maximintegrated.com/en/ds/MAX30105.pdf Page 19 Table 4
So this should be correct: https://github.com/sparkfun/SparkFun_MAX3010x_Sensor_Library/blob/d625b7e31c06d5c6a27395a46a32e6ba927b5c0b/src/MAX30105.cpp#L95

yuskegoto, 2022-11-13:
Hi guys, I got an interesting story. 

**tl;dr**
The library is not wrong. but the chip is.

I have observed there are two configurations of MAX30102 out there, (A) one with LED1 tied to Red and LED2 tied to IR, which is the right configuration as described in the datasheet, and ANOTHER version (B): LED1 and LED2 swapped for some unknown reason.
Two versions are difficult to distinguish, but pad shapes of the LED area are slightly different. Software Revision ID from two chips are also same.
I still have no clue how this kind of seemingly obvious error would happen. Maybe the config (B) did not come from Maxim factory but some kind of knock-off from salvaged chip, or maybe a custom version, but this is nothing more than my speculation. So I won't call config (B) as copy or fake.

Meanwhile with the config (B), you can still calculate heart rate, but you won't see the red LED blinking during the measurement (instead you can observe IR light with your smartphone camera). For SpO2 calculation you need to swap R and IR value. If you want to use red LED for heart rate measurement, make sure you set mode register (0x09) to "Red and IR mode"(0x03) or "Multi-LED mode"(0x07), then set setPulseAmplitudeIR() (which is actually Red), instead of R, then feed IR value to the heart beat detection, checkForBeat().

Follwing two images are above two versions of the sensors with exactly same code.
[image]
Version A: You can see the red LED is lit.

[image]
Version B: When you look carefully, you will see purple color (which indicates IR light) on the chip. We have purchased 12 pcs of them from Aliexpress, and all of them had the same configuration.

I hope this info helps you.

jens-bongartz, 2022-12-04:
Hello everyone,

we can support the above statement by [yuskegoto](https://github.com/yuskegoto).
We use the MAX30102 to measure SpO2 with our own Octave program for educational purposes. To do this, we read the RED and IR values from the sensor via getRED and getIR and plot them as graphs (see figure below).
For a (at least vital) human, the AC component of the IR channel must be larger than the AC component of the RED channel. As you can see, this is not the case with our MAX30102. It is the other way around - RED is bigger than IR. 
If we calculate Z = (AC_red / DC_red) / (AC_ir / DC_ir), we get useless numbers. If we calculate the reciprocal of Z, the numbers are correct. The only reasonable conclusion is that RED and IR are indeed reversed in our MAX30102.

[image]