sparkfun/SparkFun_MAX3010x_Sensor_Library #26: ARDUINO LIBRARY FOR MAX30101 VERSION D FOR MEASURING BLOOD PRESSURE

parminder4499, 2020-06-20:
I am using sparkfun's spo2 sensor board(MAX30101 version A) for measuring spo2 and heart rate. I want to measure blood pressure as well from MAX30101 version D which has firmware by MAXIM integrated for measuring BP. I hereby wanted to ask you if you can modify the Arduino library for MAX30101 version A, so that it can also measures Blood pressure when using version D of the sensor? 
If that library is build up then we can freely use MAX30101's different versions for making products using sparkfun's library.
Thanks

wt8008, 2020-06-24:
I assume you are referring to the MAXREFDES220. That system uses a different architecture this this board. It uses the MAX32664 to do the algorithm processing. This library directly runs an algorithm that is different from the version A and version D (of the MAX32664).

parminder4499, 2020-06-24:
yes i am referring to MAXREFDES220, the SparkFun SEN-15219 Pulse Oximeter & Heart Rate Sensor (which i am using) also uses the MAX32664 for processing( https://www.sparkfun.com/products/15219), here also MAX32664 bio sensor hub is doing the algorithm processing and i am reading the processed values using this library through I2C interface of microcontroller. however version A of the MAX32664 has different algorithm than version D of MAX32664 which includes transient Blood pressure measurement. So i guess this library wont work for reading the values from version D of MAX32664.

santaimpersonator, 2020-06-24:
@parminder4499 This library is for the MAX3010x sensors themselves. Not the MAX32664 sensor hub, which you can find in a [different GitHub repository](https://github.com/sparkfun/SparkFun_Bio_Sensor_Hub_Library). I believe your issue should be transferred to that repository.

However, since we only use the A version of the sensor hub, I am doubtful that we will do any extra development unless we adopt that hardware version. Primarily because there would be no way to do any testing/verification on our end.

parminder4499, 2020-06-25:
@santaimpersonator I got your point, Thanks for the clarification. Although I have already posted the issue on that GitHub repository. The only aim of using version D is to calculate BPT along with SPo2 and HR. I hope sparkfun will come up with non invasive BP measuring hardware design in healthcare domain. 
Thanks and Regards..