sparkfun/SparkFun_MAX3010x_Sensor_Library #31: Library doesn't work on Sparkfun ESP32 Thing Plus

theotherjenkutler, 2020-11-29:
Hardware:

Board: Sparkfun ESP32 Thing Plus
Core Installation version: 1.0.5-rc2
IDE name: Arduino IDE 1.8.13
Flash Frequency: Both 40Mhz and 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Windows 10
Description:

ESP32 never initializes an instance of this I2C device. The light on the device doesn't turn on and it doesn't appear to communicate. When tested with the exact same sketch with a Teensy LC or Sparkfun Pro Micro, it functions perfectly.
Sketch: (leave the backquotes for code formatting)

`
/*
  MAX30105 Breakout: Output all the raw Red/IR/Green readings
  By: Nathan Seidle @ SparkFun Electronics
  Date: October 2nd, 2016
  https://github.com/sparkfun/MAX30105_Breakout

  Outputs all Red/IR/Green values.

  Hardware Connections (Breakoutboard to Arduino):
  -5V = 5V (3.3V is allowed)
  -GND = GND
  -SDA = A4 (or SDA)
  -SCL = A5 (or SCL)
  -INT = Not connected

  The MAX30105 Breakout can handle 5V or 3.3V I2C logic. We recommend powering the board with 5V
  but it will also run at 3.3V.

  This code is released under the [MIT License](http://opensource.org/licenses/MIT).
*/

#include <Wire.h>
#include "MAX30105.h"

MAX30105 particleSensor;

#define debug Serial //Uncomment this line if you're using an Uno or ESP
//#define debug SerialUSB //Uncomment this line if you're using a SAMD21

void setup()
{
  debug.begin(115200);
  debug.println("MAX30105 Basic Readings Example");

  // Initialize sensor
  if (particleSensor.begin() == false)
  {
    debug.println("MAX30105 was not found. Please check wiring/power. ");
    //while (1);
  }

  particleSensor.setup(); //Configure sensor. Use 6.4mA for LED drive
}

void loop()
{
  debug.print(" R[");
  debug.print(particleSensor.getRed());
  debug.print("] IR[");
  debug.print(particleSensor.getIR());
  debug.print("] G[");
  debug.print(particleSensor.getGreen());
  debug.print("]");

  debug.println();
}
`
Debug Messages:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10900
load:0x40080400,len:6388
entry 0x400806b4
MAX30105 Basic Readings Example
MAX30105 was not found. Please check wiring/power.

EspenP, 2021-01-22:
Was this issue ever resolved? I am having a similar problem on my esp32. What's more, is I was able to detect the device with a simple i2c scan, but running the example scripts still results in the ` MAX30105 was not found. Please check wiring/power.` error.

theotherjenkutler, 2021-01-25:
No not a peep. I never managed to get mine working. Starting to explore other libraries and other ESP32 / MAX30101 and/or 30102 modules.

theotherjenkutler, 2021-01-25:
I just checked it with an Arduino Micro and this is definitely specific to ESP-32 boards. Not just the Sparkfun board but I just tested it with the ESP32 DEVKITV1. Both have ESP32-WROOM-32 processors. I'm wondering if the conflict might be with the ESP32 Board library. Might try posting there instead since there is no response from the Sparkfun folks.

theotherjenkutler, 2021-01-25:
Actually, Adafruit beat you guys to it: https://github.com/espressif/arduino-esp32/issues/4729
This is why the Sparkfun Max library doesn't work with ESP32 boards.

ROdoka, 2021-03-26:
nice, but what do I do now?
I want to use the sensor ...

I2C scanner sketch does detect it at 0x57, but the begin fails.

I don't see how to fix it and there are not any working replacement libs that I can see or find

theotherjenkutler, 2021-03-27:
Yes. The issue is with the compatibility between the ESP32 library and the Sparkfun MAX3010x library. I believe that it can be addressed on either side. It doesn't look like Sparkfun plans to maintain this library so it's probably up to Esspressif developers or maybe Adafruit will release something. It's a shame. I bought all Sparkfun products specifically to use with this library to ensure compatibility. @nseidle Is there anything you can do for us?

theotherjenkutler, 2021-03-27:
@ROdoka I just checked the ESP-32 repository and it looks like they updated some of the i2c core communication code in the last release (hopefully). I'm not sure if it will address this particular problem but it's worth a shot. Have you updated to the latest version of the ESP-32 board? I don't have time to test it tonight....

ROdoka, 2021-03-29:
@theotherjenkutler I just upgraded to 1.0.6, I see no I2C related changes in the changelog though and after compiling: same result :/ "MAX30105 was not found"

ROdoka, 2021-03-29:
I just switched to Arduino Uno and the sensor works more or less
The example with SPO2 und HR shows wrong heart rate though, it varies between 125 and 250 (while it is about 73 in reality), SPO2 looks real, though

theotherjenkutler, 2021-04-02:
@ROdoka Yes, I've had the same experience. This problem is specific to ESP32 and ESP8266.

williamesp2015, 2021-10-01:
It might help if    boolean MAX30105::begin(TwoWire &wirePort, uint32_t i2cSpeed, uint8_t i2caddr,uint8_t i2caddr = MAX30105_ADDRESS)  equal in MAX30105.cpp  and its header and laso check your SDA and SL pins and correct it i the begin function like Wire.bein(SDA,SCL); AND MOST IMPORTANT CHECK YOUR  ledMode = 3;/FOR max30101