sparkfun/SparkFun_MAX3010x_Sensor_Library #33: CAN MAX30102 DO THE SPO2, HEARTBEAT AND TEMPERATURE AT THE SAME TIME?

uyennguyen130599, 2021-03-17:
I'm doing a project with MAX30102 to detect SPO2, heartbeat and temperature at the same time. In the example of lib if I run it separately it will run as normal, but when I combine that two example, the heartbeat always 0. Can **anyone** help me with this?
Here is the code: 

#include <Wire.h>
#include "MAX30105.h"
#include "heartRate.h"

MAX30105 particleSensor;

const byte RATE_SIZE = 4; //Increase this for more averaging. 4 is good.
byte rates[RATE_SIZE];    //Array of heart rates
byte rateSpot = 0;
long lastBeat = 0; //Time at which the last beat occurred

float beatsPerMinute;
int beatAvg;
float temperature;

void setup()
{
    Serial.begin(115200);
    Serial.println("Initializing...");

    // Initialize sensor
    if (!particleSensor.begin(Wire, I2C_SPEED_FAST)) //Use default I2C port, 400kHz speed
    {
        Serial.println("MAX30105 was not found. Please check wiring/power. ");
        while (1)
            ;
    }
    Serial.println("Place your index finger on the sensor with steady pressure.");

    particleSensor.setup();                    //Configure sensor with default settings
    particleSensor.setPulseAmplitudeRed(0x0A); //Turn Red LED to low to indicate sensor is running
    particleSensor.setPulseAmplitudeGreen(0);  //Turn off Green LED
}

void loop()
{
    temperature = particleSensor.readTemperature();
    long irValue = particleSensor.getIR();
    
    if (checkForBeat(irValue) == true)
    {
        //We sensed a beat!
        long delta = millis() - lastBeat;
        lastBeat = millis();

        beatsPerMinute = 60 / (delta / 1000.0);

        if (beatsPerMinute < 255 && beatsPerMinute > 20)
        {
            rates[rateSpot++] = (byte)beatsPerMinute; //Store this reading in the array
            rateSpot %= RATE_SIZE;                    //Wrap variable

            //Take average of readings
            beatAvg = 0;
            for (byte x = 0; x < RATE_SIZE; x++)
                beatAvg += rates[x];
            beatAvg /= RATE_SIZE;
            
        }
    }
    
    Serial.print("IR=");
    Serial.print(irValue);
    Serial.print(", BPM=");
    Serial.print(beatsPerMinute);
    Serial.print(", Avg BPM=");
    Serial.print(beatAvg);
    
    Serial.print(", temperatureC=");
    Serial.print(temperature);

    if (irValue < 50000)
        Serial.print(" No finger?");

    Serial.println();
}

Okayler, 2023-10-04:
@tusharshawarma it doesnt helped. Because it was asked to read hearrate + temperature rate at the same time.... You had gave an example for heartrate + SPo2 readings ....

luoyanpei, 2023-11-07:
> I'm doing a project with MAX30102 to detect SPO2, heartbeat and temperature at the same time. In the example of lib if I run it separately it will run as normal, but when I combine that two example, [...]

> I'm doing a project with MAX30102 to detect SPO2, heartbeat and temperature at the same time. In the example of lib if I run it separately it will run as normal, but when I combine that two example, [...]

I can help you to solve this qusetion.
In the source code, the author calculates the BPM from the time difference of the board. So maybe you wrote the delay() function in your code, which caused the board time difference to be too large, and the pulse was a weak signal, so it approximated the BPM to 0