sparkfun/SparkFun_MAX3010x_Sensor_Library #34: Wrong SpO2 values. How to solve it?

AhmedAymanNegmm, 2021-05-21:
Hi

I am using example 8 on my MAX30102 to detect oxygen level, but I am getting very wrong numbers. values range in 10-100 and sometimes I get -999. Did anyone face similar issue and was able to resolve???

ASHAHEEN404, 2021-07-04:
did you found a solution?  I'm getting the same wrong values.

hishd, 2021-08-22:
I'm also facing the same issue with the MAX30102 sensor

FigsnApples, 2023-12-23:
Was there ever a resolve here... ?

hishd, 2023-12-23:
@FigsnApples I've fixed it. Please refer the below repo. I've explained it and I also attached the reference to the fix as well.

https://github.com/hishd/ESP8266_Pulse_Oximeter

FigsnApples, 2024-01-02:
> @FigsnApples I've fixed it. Please refer the below repo. I've explained it and I also attached the reference to the fix as well.
> 
> In my scenario, it was due to a cheap Chinese MAX3010 sensor
> 
> https://github.com/hishd/ESP8266_Pulse_Oximeter

Have a simplified code analogous to basic readings example of MAX30101 library? 

I am using Sparkfun's Max30101 - same deal with knockoffs on Sparkfun devices too?