sparkfun/SparkFun_MAX3010x_Sensor_Library #36: GY-MAX30100 Example8_SPO2 issue

simtcrom, 2021-09-01:
Hello,

My sensor model is GY-MAX30100 picture attached.
I am trying Example8_SPO2, and in the serial monitor I get below,

 _ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld_

I know the sensor works, Example1_Basic_Readings, Example4_HeartBeat_Plotter, Example5_HeartRate worked.
Any idea how to fix it ? 

[image]
[image]

nikkopt, 2022-11-05:
I have the exact same problem.
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
```

nikkopt, 2022-11-06:
Since there are so many articles online of people having no problems, i fired up my old virtual machine (with arduino 1.8.5, ESP version 1.0.4 and SparkFun's MAX3010x library version 1.0.8) and it works fine. Don't know which one is related to the fault since i don't have time to test right now.

nikkopt, 2022-11-06:
The culprit are the espressif libraries. The last working version is 2.0.2. Newer versions won't work.