sparkfun/SparkFun_MAX3010x_Sensor_Library #59: Watchdog triggering and getting no data.

muntakim181, 2024-03-14:
I uploaded example 8 and getting this.

`rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
`
I used esp32 doit devkit v1

alibustami, 2024-05-06:
I am having the same issue, using esp8266, the watchdog timer keeps triggering and the board resets.

leybme, 2025-03-29:
> I uploaded example 8 and got this.
> 
> `rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) configsip: 0, SPIWP:0xee clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 mode:DIO, clock div:2 load:0x3fff0030,len:1184 load:0x40078000,len:13232 load:0x40080400,len:3028 entry 0x400805e4 ` I used esp32 doit devkit v1

Remove the pluseLed pin, it is a strapping in esp32 so it became reset loop.